C0rn3j/sc-controller #15: Package for Flatpak

C0rn3j, 2024-09-01:
AppImage is pain, let's package this for Flatpak instead.

https://docs.flatpak.org/en/latest/first-build.html

Currently it seems we can't even run the Ubuntu 24.04 image on Debian 12 due to older glibC.

```bash
# ldd --version
ldd (Debian GLIBC 2.36-9+deb12u8) 2.36

# ldd --version
ldd (Ubuntu GLIBC 2.39-0ubuntu8.2) 2.39

 # ./sc-controller-v0.4.8.20-noble-x86_64.AppImage 
APPRUN_ERROR: No such file or directory
```

git-developer, 2024-09-07:
I agree that AppImage is pain. Issues are hard to tackle. The reported error is not glibc related IMHO, it's caused by differences in the directory structure between Jammy (22.04) and Noble (24.04). I already fixed this locally. Unfortunately, the Noble AppImage causes a segmentation fault when run, even when the host is Noble, too. I'm currently setting up automated tests for the AppImage build, running on several host OSs. I'll open a PR once this is ready for review.

Switching to flatpak might be an alternative to all this hassle. Actually, I can't tell because I never touched it.

git-developer, 2024-09-28:
The reported error should be fixed now. Nevertheless Flatpak might be an interesting alternative for AppImage.