C0rn3j/sc-controller #18: Enable OSD on Wayland

LynXDTS, 2024-09-08:
Please, if possible, implement the feature present in https://github.com/git-developer/sc-controller - fork of original package from kozec.

Feature in question - possibility to summon OSD in wayland.

[image]

LynXDTS, 2024-09-08:
As I see - both of the projects utilize python3 instead of original's python2

C0rn3j, 2024-09-08:
There's no OSD differences as far as I see.

Do you have the `gtk-layer-shell` dependency installed? It won't work without it.  
I just marked it and the rest of the deps as hard deps as not having it just bit me on my own system, at least in the readme, let's have a unified experience.  
It might be a good idea that the deps are verified in the code too.

Just summoning and closing the OSD menu makes the daemon sad with an error - which probably should be marked as debug instead:

```
D SCCDaemon     STICK locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.A locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.B locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
E OSD osd.menu      Sucessfully locked input
D SCCDaemon     SCButtons.B unlocked
D SCCDaemon     STICK unlocked
D SCCDaemon     SCButtons.A unlocked
```

The bigger problem is that attempting to use the "Display Keyboard" option crashes the scc-osd-daemon:

```
D SCCDaemon     STICK unlocked
D SCCDaemon     SCButtons.A unlocked
D SCCDaemon     SCButtons.B unlocked
D OSD osd.keyboard  Using font DejaVu Sans
I OSD profile       Converted button(Keys.KEY_LEFTSHIFT) to trigger(50, button(Keys.KEY_LEFTSHIFT))
I OSD profile       Converted button(Keys.KEY_LEFTCTRL) to trigger(50, button(Keys.KEY_LEFTCTRL))
D OSD Mapper        Creating virtual devices
D OSD Mapper        Keyboard: <scc.uinput.Keyboard object at 0x775af0624590>
D OSD Mapper        Mouse:    <scc.uinput.Mouse object at 0x775af06b7da0>
D OSD Mapper        Gamepad:  <scc.uinput.Dummy object at 0x775af06b57f0>
D SCCDaemon     LEFT locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     RIGHT locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     STICK locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.STICKPRESS locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.RPADTOUCH locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.LPADTOUCH locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.RPAD locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.LPAD locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.RGRIP locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.LGRIP locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.START locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.C locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.BACK locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.A locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.X locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.B locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.Y locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.LB locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.RB locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.LT locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.RT locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.CPADTOUCH locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.CPADPRESS locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.STICKPRESS locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.RSTICKPRESS locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.DOTS locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.RGRIP2 locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
D SCCDaemon     SCButtons.LGRIP2 locked by <scc.sccdaemon.Client object at 0x7fc92d7697c0>
W Actions       Action KeyboardAction can't handle button release event
D SCCDaemon     SCButtons.CPADPRESS unlocked
D SCCDaemon     SCButtons.STICKPRESS unlocked
D SCCDaemon     SCButtons.RSTICKPRESS unlocked
D SCCDaemon     SCButtons.DOTS unlocked
D SCCDaemon     SCButtons.RGRIP2 unlocked
D SCCDaemon     SCButtons.LGRIP2 unlocked
D SCCDaemon     LEFT unlocked
D SCCDaemon     RIGHT unlocked
D SCCDaemon     STICK unlocked
D SCCDaemon     SCButtons.STICKPRESS unlocked
W SCCDaemon     scc-osd-daemon died; restarting after 5s
D SCCDaemon     SCButtons.RPADTOUCH unlocked
D SCCDaemon     SCButtons.LPADTOUCH unlocked
D SCCDaemon     SCButtons.RPAD unlocked
D SCCDaemon     SCButtons.LPAD unlocked
D SCCDaemon     SCButtons.RGRIP unlocked
D SCCDaemon     SCButtons.LGRIP unlocked
D SCCDaemon     SCButtons.C unlocked
D SCCDaemon     SCButtons.START unlocked
D SCCDaemon     SCButtons.BACK unlocked
D SCCDaemon     SCButtons.A unlocked
D SCCDaemon     SCButtons.X unlocked
D SCCDaemon     SCButtons.B unlocked
D SCCDaemon     SCButtons.Y unlocked
D SCCDaemon     SCButtons.LB unlocked
D SCCDaemon     SCButtons.RB unlocked
D SCCDaemon     SCButtons.LT unlocked
D SCCDaemon     SCButtons.RT unlocked
D SCCDaemon     SCButtons.CPADTOUCH unlocked
I SCCDaemon     scc-osd-daemon lost
D OSD DaemonCtrl    Connected to daemon, version 0.4.8.20
D OSD DaemonCtrl    Daemon reported profile change for sc1: /usr/share/scc/default_profiles/XBox Controller with High Precision Camera.sccprofile
D OSD DaemonCtrl    Daemon is ready.
I SCCDaemon     Registered scc-osd-daemon
I OSD osd.daemon    Sucessfully registered as scc-osd-daemon
```

Tested on a Plasma Wayland session (so kwin).

LynXDTS, 2024-09-08:
Just spun up a fresh Arch VM, as I do not remember all of the configuration on main system.

Installed Yay, got sc-controller-git from AUR, installed gtk-layer-shell explicitly after that and passed my controller through - still no OSD upon pressing the "Steam button", otherwise app is functional. Tried to just launch sc-controller from terminal - no relevant errors it seems, do not know how do debug deeper than that, sorry.

C0rn3j, 2024-09-08:
kill sc-controller, kill scc-daemon, launch `scc-daemon debug` in one terminal and `sc-controller` in another.

I am also on Arch.  
What compositor did you try?

LynXDTS, 2024-09-08:
Upon reboot magically fixed itself it seems, idk what could have been a problem. 

Answering the question - KWin as well, I assume.

C0rn3j, 2024-09-08:
If you updated the package you might have pulled `python-setuptools` on top of it, it was missing in the -git package, but it should *only* have been a dependency for a fully functional About menu.

Glad it works now.  
Leaving the issue open because the OSD is still partially broken.

C0rn3j, 2024-09-17:
https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/1141

https://gitlab.gnome.org/GNOME/mutter/-/issues/973

Apparently GNOME's mutter refuses to support Layer Shell, so it just won't work™ there I suppose unless someone puts in the necessary work to figure out a solution.

Skaifer, 2024-12-01:
OSD not working on my end as well. I'm running an appimage Jammy version of SC controller. PikaOS, KDE, Wayland. I've installed libgtk-layer-shell0 0.8.2-1, which seems to be latest Debian based package, and menus do not appear at all. Also, mouse region isn't working. That is mentioned there. I just wonder, does mouse region work on Wayland through Steam input as well, or it's just me problem?

robohappy, 2025-05-13:
Tested with Wayland. Menu and Gui shows and program works great. However when launching a game and trying to access the OSD, the menu renders behind the game, so it is not visible when trying to switch layouts when gaming. Is there a fix or workaround for this in Wayland?

robohappy, 2025-05-31:
Checked over at the gtk-layer-shell github page for a possible solution.

Is sc-controller using the GTK_LAYER_SHELL_LAYER_OVERLAY with gtk-layer-shell?

robohappy, 2025-09-10:
Following up on this issue, in an older post you mentioned mutter not supporting the necessary extension. Does KDE Wayland experience the same issue, or could something be implemented to allow the menu to always show over the game window for KDE Wayland, instead of rendering behind the game window?

C0rn3j, 2025-09-10:
Plasma supports it.

> could something be implemented to allow the menu to always show over the game window for KDE Wayland, instead of rendering behind the game window

I think we'd need something like xdg-pip for that?

https://gitlab.freedesktop.org/wayland/wayland-protocols/-/issues/44

C0rn3j, 2026-08-16:
Switch To now no longer crashes scc-daemon under Wayland, it has the same behavior as Autoswitcher (informs it is not supported under Wayland).

Does anyone actually use the Switch To feature, though?

Switch To gives you a list of application Windows and lets you give them focus.

It was an easy thing to do on X11, but it is compositor specific on Wayland, there does not seem to be any unified protocol for it, and it is basically just a reimplemented ALT+TAB menu, which could be achieved by binding... ALT+TAB.

Basically it seems like a lot of trouble for a little gain, and it is even disabled by default - discoverability is poor - so I presume people aren't even using it and it'd be reasonable to just drop it.