C0rn3j/sc-controller #21: Trigger Doesn't Output Analog Value

0-ks, 2024-09-16:
It's not possible to emulate Analog trigger values.

As you can see in the Wine Control Panel screenshot bellow, User 0 is the emulated device by SC Controller and the User 1 is the direct output of the DS4 Controller before going through SC Controller processing:
[image]

Here is the output of the emulated device in EVTEST to confirm the issue is not related to Wine:
[image]

After further investigation, I found that changing the Analog Output Range Start and End to 128 and 129 resolves the problem, so it seems like the default range of 0 to 255 is causing the problem for some reason.

As a workaround, it is possible to set:
`axis(Axes.ABS_Z, 128, 129)` for the Left Trigger Action
and
`axis(Axes.ABS_RZ, 128, 129)` for the Right Trigger Action
until the bug is fixed.

C0rn3j, 2024-09-16:
How can I repro this?

I got as far as running `wine control`, connecting a clone PS4 controller over USB and emulating the default XBox Controller profile, which works fine.

[Screencast_20240916_145933.webm](https://github.com/user-attachments/assets/049ea69e-91fe-4f11-997f-73c598a4e198)

Note that due to https://github.com/C0rn3j/sc-controller/issues/13#issuecomment-2336750409 I am using Evdev instead of HIDRAW which is completely broken for me.

0-ks, 2024-09-16:
> How can I repro this?
> 
> I got as far as running `wine control`, connecting a clone PS4 controller over USB and emulating the default XBox Controller profile, which works fine.
> Screencast_20240916_145933.webm
> 
> Note that due to [#13 (comment)](https://github.com/C0rn3j/sc-controller/issues/13#issuecomment-2336750409) I am using Evdev instead of HIDRAW which is completely broken for me.

Sometimes Wine doesn't detect SC-Controller's output as an XInput device and only shows the output directly from the device itself, In that case you can check the DInput tab to see the axes values for each device, make sure to select "Controller (XBOX 360 For Windows)" from the list in the Input tab:

[image]

C0rn3j, 2024-09-16:
> make sure to select "Controller (XBOX 360 For Windows)" from the list in the Input tab

I *only* have that option, are you connecting two controllers?

0-ks, 2024-09-16:
> > make sure to select "Controller (XBOX 360 For Windows)" from the list in the Input tab
> 
> I _only_ have that option, are you connecting two controllers?
> 
> I am trying this on wine-staging 9.10.

No, Just a single DS4 with Bluetooth, But it's not a DS4 related issue as I first encountered the problem when I was trying to remap my EdgeTX Radio Controller using SC-Controller.

C0rn3j, 2024-09-16:
Aha, I presume you can't repro this over USB?

It's broken in BT and works over USB for me with DS4.

C0rn3j, 2024-09-16:
Looks messed up and my right controller trigger does not even register at all over BT in the first place.

[image]

[image]

[image]

[image]

What's your PS4 VID/PID from lsusb?

`Bus 005 Device 031: ID 054c:09cc Sony Corp. DualShock 4 [CUH-ZCT2x]`

0-ks, 2024-09-16:
> Looks messed up and my right controller trigger does not even register at all over BT in the first place.

Sometimes Both the Right and the Left triggers values are shown in Z axes for some reason and the Rz axes is not showing any value at all, but for me both triggers still work, it just seems like the Z axes shows the sum of the both trigger values, but other times Z shows the left trigger and Rz the right trigger.

> What's your PS4 VID/PID from lsusb?

Mine is:
`054c:05c4 Sony Corp. DualShock 4 [CUH-ZCT1x]`

> What's your kernel version? uname -r -> Apparently it matters https://github.com/kozec/sc-controller/issues/262

I'm on ArchLinux and my kernel is `6.10.9-zen1-2-zen` and the DS4 controller works flawlessly in every other software.

But I don't think it's related to the controller or a specific kernel, since I can reproduce the bug with other controllers as well.

C0rn3j, 2024-09-16:
The problem seems to be that whatever syspath sc-controller gets is mangled, i.e. `/proc/bus/input/devices` and dmesg have  
`Sysfs=/devices/virtual/misc/uhid/0005:054C:05C4.002E/input/input180`  
but sc-controller checks in  `/sys/bus/hid/devices` and gets
`/sys/devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-7/3-7:1.0/bluetooth/hci0/hci0:50`  
which does not have the necessary event* links for whatever reason which it expects to find, it's a bit low level for me.

The fact that majority of the codebase isn't typed and I have to dig through things every time I try to find how something works isn't helping.

C0rn3j, 2024-09-16:
Yeah soo, that's ANOTHER issue.

```
	def make_evdev_device(sys_dev_path: str, *whatever):
		sys_dev_path = '/sys/bus/hid/devices/0005:054C:05C4.0036' # HARDCODE TEST
```

This indeed gets it to detect my controller properly and I am at a loss where the damn syspath comes from, looks like it might be the libinput C binding but it's really a bit beyond me.

Now that it's detected properly, triggers get detected but they're as broken as described in your OP.

Do you have no issues with the PS4 BT sc driver picking your controller up? All connects and you see the PS icons while not having to define the controller as a custom device?

Could you post dmesg from controller connect if so?

I always have it fail like this without the brutal hack:

```
W DS4           Failed to determine controller device 'None' for '/sys/devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-7/3-7:1.0/bluetooth/hci0/hci0:50'
```

Because `devices = get_evdev_devices_from_syspath(sys_dev_path)` just finds nothing on that cursed syspath.

0-ks, 2024-09-16:
> Do you have no issues with the PS4 BT sc driver picking your controller up? All connects and you see the PS icons while not having to define the controller as a custom device?

In general I don't have any problems using it over BT, Steam detects it correctly and even when I'm not using Steam Input, for example running games in Bottles with Wine all the games detect the controller and I don't have any issues,

But for SC-Controller, I had to add it in the Controllers Tab > Register New Controller, the same is true about my EdgeTX Radio Controller, but when connecting the DS4 over USB, SC-Controller detects it automatically with correct labels.

> Could you post dmesg from controller connect if so?

Sure, Here is my dmesg right after I connect the DS4 over Bluetooth:

```
[223244.839084] playstation 0005:054C:05C4.0031: unknown main item tag 0x0
[223244.839868] playstation 0005:054C:05C4.0031: hidraw9: BLUETOOTH HID v1.00 Gamepad [Wireless Controller] on xx:xx:xx:xx:xx:xx
[223244.855830] input: Wireless Controller as /devices/virtual/misc/uhid/0005:054C:05C4.0031/input/input181
[223244.856043] input: Wireless Controller Motion Sensors as /devices/virtual/misc/uhid/0005:054C:05C4.0031/input/input182
[223244.856134] input: Wireless Controller Touchpad as /devices/virtual/misc/uhid/0005:054C:05C4.0031/input/input183
[223244.856318] playstation 0005:054C:05C4.0031: Registered DualShock4 controller hw_version=0x00004302 fw_version=0x00000054
```

C0rn3j, 2024-09-16:
> But for SC-Controller, I had to add it in the Controllers Tab > Register New Controller

Ah, I thought it worked for you for some reason, when https://github.com/C0rn3j/sc-controller/issues/22 gets fixed (somehow), it'll show up proper.