C0rn3j/sc-controller #24: `KeyError: 'KEY_ESC'` when starting daemon in 0.4.9.0.1-bullseye

git-developer, 2024-09-21:
```
# ./sc-controller-v0.4.9.0.1-bullseye-x86_64.AppImage daemon debug
Traceback (most recent call last):
  File "/tmp/.mount_sc-conhVv3zZ/usr/bin/scc-daemon", line 2, in <module>
    from scc.sccdaemon import SCCDaemon
  File "/tmp/.mount_sc-conhVv3zZ/usr/lib/python/scc/sccdaemon.py", line 10, in <module>
    from scc.uinput import CannotCreateUInputException
  File "/tmp/.mount_sc-conhVv3zZ/usr/lib/python/scc/uinput.py", line 65, in <module>
    Keys["KEY_ESC"]: 0x70029,
  File "/tmp/.mount_sc-conhVv3zZ/usr/lib/python3.9/enum.py", line 408, in __getitem__
    return cls._member_map_[name]
KeyError: 'KEY_ESC'
```

C0rn3j, 2024-09-21:
The fun here is twofold, I torn out the ultraancient enum included in the project, and enum also had major unifying changes in 3.11, which I presume is what would start working (I test on 3.12 and pray my linter does a good job for older versions) trying to look into this now.

https://docs.python.org/3/whatsnew/3.11.html#enum

git-developer, 2024-09-21:
I'm sorry that I can't help you with that stuff, but Python is not my main business...

> I test on 3.12 and pray my linter does a good job for older versions

Well, erm, you may do so of course, but I wouldn't expect that the result works on older distros. It should be safer to develop and test in the oldest supported environment. Newer versions are often backwards-compatible with older ones, but not the other way round. But that's just my 2ct.

C0rn3j, 2024-09-21:
Absolutely, I will probably switch to the dockerized version built against the oldest supported version at some point, otherwise I'll just keep introducing more breaking changes, hoping that the linter would catch everything turned out to work not so good :)

git-developer, 2024-09-21:
I hope you noticed https://github.com/C0rn3j/sc-controller/pull/26 which resolves the crash for me. Probably no long-term solution, but might help in the investigation.

closed 2024-09-21 by C0rn3j

git-developer, 2024-09-22:
> I test on 3.12

One more idea: we could run the scc build on every supported python version, like [this](https://github.com/git-developer/sc-controller/blob/b8fc03291fbb7eca119c112a3b13c2326bb8d396/.github/workflows/scc-linux.yml#L21). Do you think this would improve the situation? I could open another PR then. Builds [fine](https://github.com/git-developer/sc-controller/actions/runs/10978190088), but I think the tests currently don't cover all the compatibility issues.

C0rn3j, 2024-09-22:
I think building it once on the oldest and once on the latest might catch something, but I don't think interim versions between those would reveal anything else, so it sounds like having 3.8 and 3.12 would be a good idea?

I have for now at least made run.sh infinitely less annoying to use and infinitely more reliable and spun up an old Debian container to at least test the CLI parts in.