C0rn3j/sc-controller #27: cannot invert CEMU Y-Axis

UltraBlackLinux, 2024-09-21:
I've been running into this issue where I'm not able to change the sensitivity of the gyro with this config:
`mode(LPAD, None, sens(-5.0, 5.0, gyro(Axes.ABS_Y, Axes.ABS_X)))`
I can neither invert their axes, nor can I change the sensitivity. None of that has any effect.
(I would also like to mention that whenever I re-open the gyro menu all my settings are reset to the defaults and I have to re-enter everything, quite frustrating...)
Any idea what's going on?
I'm on v0.4.8.21 using a Steam Controller.
Thanks!

C0rn3j, 2024-09-21:
How can I test and reproduce this? Never used/debugged the gyro before.

Also please try testing on beta (currently v0.4.9.1.1) too in case it happens to be fixed there.

UltraBlackLinux, 2024-09-21:
just tried the beta. It doesn't seem to be completely resetting everything like it was previously, but the sensitivity settings are still quietly discarded, so that's still not working.

C0rn3j, 2024-09-21:
How is your Steam Controller connected? BT, Dongle, USB?

Which site are you using to test?

Which profile are you loading on SC Controller while testing?

I can't get https://hardwaretester.com/gamepad to see any axis movement on neither USB DS4v2 nor Dongle SC on the XBox+Camera profile.

UltraBlackLinux, 2024-09-21:
hmm well I was using that exact website. SC connected via USB.
that website doesn't seem to do the gyro natively, but if you map the gyro to a stick, like with my above example, you can see the movements perfectly fine.

C0rn3j, 2024-09-21:
[image]

Right, got it, just going to leave that here, this project could really use some feature demo video.

I can repro it.  
No clue why it happens, will check it out.

C0rn3j, 2024-09-21:
You *can* actually set the Axis invert, but when you open the menu again, it loads it without it, and pressing OK overwrites the setting.

The custom action sens setting don't even save once though.

C0rn3j, 2024-09-22:
Try v0.4.9.2, it's still fairly broken in the Custom Action menu, but as long as you set that first and set sensitivty after, it'll work.

UltraBlackLinux, 2024-09-22:
that seems to have fixed most of the woes there! As you correctly pointed out there's still something weird going on with the custom action tab, but other than that it's working great now, thank you so much!

commit 9f7dc8b6 (2024-09-22): Gyro action: If Action is a bare SensitivityModifier, set speeds in the UI for it too, partially fixes #27