C0rn3j/sc-controller #38: Opening the About dialog causes `No module named 'pkg_resources'`

git-developer, 2024-10-04:
```
Traceback (most recent call last):
  File "/tmp/.mount_sc-congOmcol/usr/lib/python3.10/site-packages/scc/gui/app.py", line 870, in on_mnuAbout_activate
    AboutDialog(self).show(self.window)
  File "/tmp/.mount_sc-congOmcol/usr/lib/python3.10/site-packages/scc/gui/aboutdialog.py", line 12, in __init__
    self.setup_widgets()
  File "/tmp/.mount_sc-congOmcol/usr/lib/python3.10/site-packages/scc/gui/aboutdialog.py", line 25, in setup_widgets
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
```

Environment: `sc-controller-v0.4.9.4-jammy-x86_64.AppImage`

The build log contains a [deprecation warning](https://github.com/C0rn3j/sc-controller/actions/runs/11168587910/job/31047323442#step:9:129) about `pkg_resources`, not sure if it's related.

C0rn3j, 2024-10-04:
[image]

I was about to get annoyed because I couldn't find it, then realized it's hanging in my uncommitted patches :)

Depends on https://github.com/C0rn3j/sc-controller/pull/30 so we don't have to keep redoing it.

C0rn3j, 2024-10-04:
Fixed by https://github.com/C0rn3j/sc-controller/commit/045de590d05cc273c5d0372ad232b37b8f8d3a63

closed 2024-10-04 by C0rn3j