C0rn3j/sc-controller #40: Connecting a steam controller causes FileNotFoundErrors

git-developer, 2024-10-04:
These messages were shown when I connected a steam controller via USB:

```
E evdev         [Errno 2] No such file or directory: '/dev/input/event24'
Traceback (most recent call last):
  File "/tmp/.mount_sc-conk3I8dd/usr/lib/python3/dist-packages/evdev/device.py", line 126, in __init__
    fd = os.open(dev, os.O_RDWR | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/input/event24'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/tmp/.mount_sc-conk3I8dd/usr/lib/python3.10/site-packages/scc/drivers/evdevdrv.py", line 398, in handle_new_device
    dev = evdev.InputDevice(eventnode)
  File "/tmp/.mount_sc-conk3I8dd/usr/lib/python3/dist-packages/evdev/device.py", line 128, in __init__
    fd = os.open(dev, os.O_RDONLY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/input/event24'
E evdev         [Errno 2] No such file or directory: '/dev/input/event23'
Traceback (most recent call last):
  File "/tmp/.mount_sc-conk3I8dd/usr/lib/python3/dist-packages/evdev/device.py", line 126, in __init__
    fd = os.open(dev, os.O_RDWR | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/input/event23'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/tmp/.mount_sc-conk3I8dd/usr/lib/python3.10/site-packages/scc/drivers/evdevdrv.py", line 398, in handle_new_device
    dev = evdev.InputDevice(eventnode)
  File "/tmp/.mount_sc-conk3I8dd/usr/lib/python3/dist-packages/evdev/device.py", line 128, in __init__
    fd = os.open(dev, os.O_RDONLY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/input/event23'
E evdev         [Errno 2] No such file or directory: '/dev/input/event22'
Traceback (most recent call last):
  File "/tmp/.mount_sc-conk3I8dd/usr/lib/python3/dist-packages/evdev/device.py", line 126, in __init__
    fd = os.open(dev, os.O_RDWR | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/input/event22'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/tmp/.mount_sc-conk3I8dd/usr/lib/python3.10/site-packages/scc/drivers/evdevdrv.py", line 398, in handle_new_device
    dev = evdev.InputDevice(eventnode)
  File "/tmp/.mount_sc-conk3I8dd/usr/lib/python3/dist-packages/evdev/device.py", line 128, in __init__
    fd = os.open(dev, os.O_RDONLY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/input/event22'
```

Environment: `sc-controller-v0.4.9.4-jammy-x86_64.AppImage`

I went to `/dev/input/` after seeing these message. The OS is right: both files `/dev/input/event23` and `/dev/input/event24` do not exist. The last devices are:
```
/dev/input/event19:     SCController Keyboard
/dev/input/event20:     SCController Mouse
/dev/input/event21:     Microsoft X-Box 360 pad
```

C0rn3j, 2024-10-04:
iirc it's some event123_whatever files which we shouldn't be matching but there's a dumb wildcard instead of a regex.

C0rn3j, 2024-10-28:
The relevant function should be here if you want to take a stab at it

```python
	@staticmethod
	def get_event_node(syspath: str):
		filename = syspath.split("/")[-1]
		if not filename.startswith("event"):
			return None
		print(syspath)
		return f"/dev/input/{filename}"
```

Just add the print in `drivers/evdevdrv.py` and run `./run.sh daemon`

It's hard to repro, managed to do it twice in about 15 minutes with a PS4 controller over USB (might have accidentally connected it over BT) as the errors were triggered by UHID paths.

We could simply do a check if `/dev/input/{filename}` exists before returning it at that point, but it'd be nicer to be able to reliably repro and see what's happening better.

git-developer, 2024-10-28:
Thanks for looking. I'm currently quit busy, I'll have a look and report back when I find the time.