C0rn3j/sc-controller #43: Migrating away from GTK3 and Glade

C0rn3j, 2024-10-04:
This is mostly a placeholder issue.

Glade is [unmaintained since 2022](https://gitlab.gnome.org/GNOME/glade) and only for GTK3.  
Also not very well liked - https://blogs.gnome.org/christopherdavis/2020/11/19/glade-not-recommended/

Aside from closing our eyes and ignoring the issue, there are two avenues we can take, both of which require us to move away from Glade:
  * Port to GTK4 - there was an attempt in upstream, but author lost interest when they found out the project is supposed to be rewritten - https://github.com/kozec/sc-controller/issues/706
  * Port to another toolkit like Qt6

Jazztache, 2025-05-18:
I know that PySide for Qt binds well to the C++ bindings. It would also integrate nicely within the Steam Deck's desktop mode.

This is because KDE itself is using Qt, and I think SC-Controller with a Breeze theme would fit in much better with the interface.

C0rn3j, 2026-09-09:
Temporary GTK3 branch that I just split off from main is now over at https://github.com/C0rn3j/sc-controller/tree/v0.7.x_gtk3

closed 2026-09-09 by C0rn3j