C0rn3j/sc-controller #44: Windows version

C0rn3j, 2024-10-04:
Upstream thread - https://github.com/kozec/sc-controller/issues/215

Upstream wiki on how to get it running - https://github.com/kozec/sc-controller/wiki/Running-SC-Controller-on-Windows

Would be nice to test it to see if/how well it works.

danielbao, 2024-11-08:
I would definitely like to test it out since I can't install steam on some of my work/uni computers and love the touchpad usage. However, on the releases I only see AppImages and I'm not sure how to compile the raw python folder into a windows app. If you could provide some pointers on that end I could try that too!

C0rn3j, 2024-11-08:
Sooo looking at this, I have no clue how this was meant to work, as even the old Python 2 kozec releases don't ship any EXEs in them.

You're going to have to install Python (and maybe git?) first.

I would follow https://github.com/C0rn3j/sc-controller?tab=readme-ov-file#via-python-into-a-local-build-directory  
You might have to setup a venv first (run.sh does that, so try that/look at how it does it).

Now the run step will be a bit different, but let's get to that when it builds.

You might have to set/play with the platforms - https://github.com/C0rn3j/sc-controller/blob/eef0e518a0041224d72d24c7010ac3e34559de4c/pyproject.toml#L23

C0rn3j, 2026-09-25:
I did get both the Python daemon and the GUI working, but I had to gut all the device emulation and USB/BT reading to do so.

OSD 99% won't work as-is either.

So barebones support is there now, over at https://github.com/C0rn3j/sc-controller/tree/windows

hifihedgehog, 2026-10-21:
@C0rn3j On the TODO item about HIDMaestro: I'm its author, so here is how it would fit, in case it saves you a look.

The SDK is a .NET 10 library, and Python can drive it through pythonnet, as [JoystickGremlinEx](https://github.com/muchimi/JoystickGremlinEx) does. That adds the .NET 10 runtime as a dependency. The `xbox-360-wired` profile gives an Xbox 360 pad with XInput through a companion device, and `dualshock-4-v2` covers the DS4 emulation item. The driver runs in user mode, signed with a certificate made on the machine, so there is no ViGEmBus, no kernel driver and no test-signing mode. The process that creates controllers runs elevated. HIDMaestro does not hide the physical controller, so HidHide stays in the picture for double input. It is MIT licensed, which works with GPL-2.0.

Glad to help if you try it.

C0rn3j, 2026-11-06:
Thank you for taking the time to respond, saving me from digging into a bit of the technical details, and offering further help.

However, I don't think I will be using that dependency, due to the religious nonsense included everywhere, from the `README` to the `License` file.