C0rn3j/sc-controller #47: Project name

C0rn3j, 2024-10-06:
Currently, the project name ranges from:

Repo name:  
* "sc-controller" -> This has to be lower-case and dash required for spaces, same for packages on some distros.  
This is irrelevant.

Readme:   
* "SC Controller" -> https://github.com/C0rn3j/sc-controller?tab=readme-ov-file#sc-controller

Wiki:   
* "SC-Controller" -> https://github.com/C0rn3j/sc-controller/wiki#sc-controller-wiki
* "SC Controller" -> https://github.com/C0rn3j/sc-controller/wiki/Running-SC-Controller-on-Windows

Skaifer, 2024-12-11:
Unprofessional opinion. I would say that the mention of the SC Controller being used with the DualShock controller is playing havoc with my brain.

nycki93, 2025-05-03:
I think SC Controller is fine. If you wanted to change its name for some reason, I think the most important things to capture are:

- compatibility with features that aren't in xinput, like gyro and touchpads
- ability to "hide" the source controller, so that you can map buttons to buttons
- similarity to Steam Input, except that it runs without Steam, and on low-power devices like a Raspberry Pi
- similar to the old name, for continuity.

I would propose something like "Any Controller" or "Sticky Controller" or something else in that style.