C0rn3j/sc-controller #50: Fix building from archived source code without .git

C0rn3j, 2024-10-06:
@git-developer Integrating setuptools-scm in #30 broke this.

We could [disable the Assets generating the source builds](https://github.com/orgs/community/discussions/6003#discussioncomment-10480237) and make everyone clone from git, but: 
 * That's not ideal
 * Regular Zipped downloads such as https://github.com/C0rn3j/sc-controller/archive/refs/tags/v0.4.9.5.zip would still exist 
 * We point to the tarballs in multiple places at the moment.

We could simply also hardcode a "unknown version from tarball" or otherwise hardcode some version, but that might make supporting it a nightmare, and it would break the daemon vs sc-controller version mismatch detection.

What we could do that'd possibly be a good solution is have the CI commit the calculated git version to the repo automatically?  
But I am unsure how to do it in a way which would actually be up to date and not `current version - 1`.

I am blanking out on any other good solutions.

```python
[0] % ./run.sh       
* Creating isolated environment: venv+pip...
* Installing packages in isolated environment:
  - setuptools-scm>=8.0
  - setuptools>=60
  - wheel
* Getting build dependencies for wheel...
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/pyproject_hooks/_in_process/_in_process.py", line 373, in <module>
    main()
  File "/usr/lib/python3.12/site-packages/pyproject_hooks/_in_process/_in_process.py", line 357, in main
    json_out["return_val"] = hook(**hook_input["kwargs"])
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pyproject_hooks/_in_process/_in_process.py", line 134, in get_requires_for_build_wheel
    return hook(config_settings)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools/build_meta.py", line 332, in get_requires_for_build_wheel
    return self._get_build_requires(config_settings, requirements=[])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools/build_meta.py", line 302, in _get_build_requires
    self.run_setup()
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools/build_meta.py", line 318, in run_setup
    exec(code, locals())
  File "<string>", line 45, in <module>
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools/__init__.py", line 117, in setup
    return distutils.core.setup(**attrs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 145, in setup
    _setup_distribution = dist = klass(attrs)
                                 ^^^^^^^^^^^^
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools/dist.py", line 283, in __init__
    _Distribution.__init__(self, dist_attrs)
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 279, in __init__
    self.finalize_options()
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools/dist.py", line 635, in finalize_options
    ep(self)
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools_scm/_integration/setuptools.py", line 123, in infer_version
    _assign_version(dist, config)
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools_scm/_integration/setuptools.py", line 58, in _assign_version
    _version_missing(config)
  File "/tmp/build-env-58yivh01/lib/python3.12/site-packages/setuptools_scm/_get_version_impl.py", line 117, in _version_missing
    raise LookupError(
LookupError: setuptools-scm was unable to detect version for /home/c0rn3j/Downloads/sc-controller-0.4.9.5.

Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.

For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj

ERROR Backend subprocess exited when trying to invoke get_requires_for_build_wheel

```

git-developer, 2024-10-06:
OK, I see. That's a design problem. We decided to have a single source of truth, and this single source is Git now. So, if you  download a tarball, that's not enough. From my point of view, we have to decide between
1. Make Git a hard requirement for a build.
    - The sources are managed in Git, so it should at least be possible to every client to access it.
    - If building from tarball/zip is a requirement, this option is out.
3. Move the single source of truth away from Git into a file.
    - The problem is, this file must be managed, for every branch. This must be done automatically somehow.

I think we don't want to create a second commit with an updated version number in the CI for each code change commit. Think of how the history would look like.

How do other projects solve this problem? Every user of `setuptools-scm` must be affected. Other projects I know actually use tarballs as archive only and always build from git.

git-developer, 2024-10-06:
There seems to be an [env variable](https://github.com/pypa/setuptools-scm/blob/main/docs/config.md#environment-variables) `SETUPTOOLS_SCM_PRETEND_VERSION_FOR_${PACKAGE_NAME}` that might be used for non-git builds. But this doesn't help in finding an appropriate value. Is it maybe possible to add some file on creation of a a source tarball?

Idea: we could manually create a source zip in the CI including the version info.

C0rn3j, 2024-10-06:
One thing we could do is detect lack of .git and try parsing the parent directory in that name case - i.e. `sc-controller-0.4.9.5`.

git-developer, 2024-10-06:
Good idea. AFAIK, this directory name should be composed of the project name and release version. So even branches / tags could be handled properly.

C0rn3j, 2026-10-19:
Ended up just treating .zip releases as unsupported from the point of having the correct version.

[image]

closed 2026-10-19 by C0rn3j