C0rn3j/sc-controller #54: ctypes.ArgumentError / TypeError: wrong type in drivers/steamdeck.py

bell07, 2024-10-27:
Currently I use sc-controller-0.4.8.17 on my SteamDeck in Gentoo Linux. the version is provided by Gentoo Repository.

Since I have some small issues with the version, I tried the Beta AppImage today.
unfortunately the daemon repeats an error in loop.

```
player@game-deck ~/Downloads $ ./sc-controller-0.4.9.6-trixie-x86_64.AppImage daemon debug
D SCCDaemon     Starting SCCDaemon...
D SCCDaemon     Initializing drivers...
W SCCDaemon     Skipping disabled driver 'ds4drv'
W SCCDaemon     Skipping disabled driver 'ds5drv'
W SCCDaemon     Skipping disabled driver 'fake'
W SCCDaemon     Skipping disabled driver 'remotepad'
D USB           Registered USB driver for 28de:1102
D USB           Registered USB driver for 28de:1142
D USB           Registered USB driver for 28de:1205
D Mapper        Creating virtual devices
D Mapper        Keyboard: <scc.uinput.Keyboard object at 0x7fd98482a690>
D Mapper        Mouse:    <scc.uinput.Mouse object at 0x7fd98482a3c0>
D Mapper        Gamepad:  <scc.uinput.UInput object at 0x7fd98482b500>
D SCCDaemon     Created control socket /home/player/.config/scc/daemon.socket
D SCCDaemon     Connected to XServer :0.0
Exception in thread Thread-2 (_threaded):
Traceback (most recent call last):
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
Exception in thread Thread-3 (_threaded):
Traceback (most recent call last):
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/threading.py", line 1012, in run
    self.run()
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/site-packages/scc/sccdaemon.py", line 1443, in _threaded
    self._target(*self._args, **self._kwargs)
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/site-packages/scc/sccdaemon.py", line 1443, in _threaded
    self.p = subprocess.Popen(self.args, stdin=None)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    self.p = subprocess.Popen(self.args, stdin=None)
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/subprocess.py", line 1026, in __init__
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/subprocess.py", line 1955, in _execute_child
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/subprocess.py", line 1955, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/player/Downloads/usr/bin/python'
FileNotFoundError: [Errno 2] No such file or directory: '/home/player/Downloads/usr/bin/python'
D asyncio       Using selector: EpollSelector
D deck          Got SteamDeck with serial 3:3
D USB           USB device added: 28de:1205
D root          TODO: Hardcoded 1s sleep!
D SCCDaemon     Assigned default_mapper to <Deck deck3:3>
D SCCDaemon     Turning gyrosensor ON
D SCCDaemon     Controller added: <Deck deck3:3>
E USB           Failed to handle received data
E USB           argument 2: TypeError: wrong type
E USB           Traceback (most recent call last):
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/site-packages/scc/drivers/usb.py", line 51, in callback_wrapper
    callback(endpoint, data)
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/site-packages/scc/drivers/steamdeck.py", line 196, in _on_input
    ctypes.memmove(ctypes.addressof(self._input), data, len(data))
ctypes.ArgumentError: argument 2: TypeError: wrong type

E USB           Failed to handle received data
E USB           argument 2: TypeError: wrong type
E USB           Traceback (most recent call last):
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/site-packages/scc/drivers/usb.py", line 51, in callback_wrapper
    callback(endpoint, data)
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/site-packages/scc/drivers/steamdeck.py", line 196, in _on_input
    ctypes.memmove(ctypes.addressof(self._input), data, len(data))
ctypes.ArgumentError: argument 2: TypeError: wrong type

E USB           Failed to handle received data
E USB           argument 2: TypeError: wrong type
E USB           Traceback (most recent call last):
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/site-packages/scc/drivers/usb.py", line 51, in callback_wrapper
    callback(endpoint, data)
  File "/tmp/.mount_sc-conLqso8q/usr/lib/python3.12/site-packages/scc/drivers/steamdeck.py", line 196, in _on_input
    ctypes.memmove(ctypes.addressof(self._input), data, len(data))
ctypes.ArgumentError: argument 2: TypeError: wrong type

[ ... repeat the TypeError .... ]
```

C0rn3j, 2024-10-28:
Try .7, I have guessed at the issue.

New images should be built in a couple minutes - https://github.com/C0rn3j/sc-controller/releases/tag/v0.4.9.7

Reopen if it persists. 

The `$(pwd)/usr/bin/python` problemis still there, but it does not seem to be fatal.

closed 2024-10-28 by C0rn3j

bell07, 2024-10-28:
Thank you! The 0.4.9.7 does work on SteamDeck now again

commit 0d315214 (2024-10-28): Steam Deck: Attempt to guess an USB issue to fix #54