C0rn3j/sc-controller #55: 0.4.8.21 System sleep crash

bell07, 2024-10-27:
If I try to set my SteamDeck into sleep (or hibernate) after resume the sc-controller crashes

```
player@game-deck ~/Downloads $ ./sc-controller-v0.4.8.21-bookworm-x86_64.AppImage daemon debug
D SCCDaemon     Starting SCCDaemon...
D SCCDaemon     Initializing drivers...
W SCCDaemon     Skipping disabled driver 'ds4drv'
W SCCDaemon     Skipping disabled driver 'ds5drv'
W SCCDaemon     Skipping disabled driver 'fake'
W SCCDaemon     Skipping disabled driver 'remotepad'
D USB           Registered USB driver for 28de:1102
D USB           Registered USB driver for 28de:1142
D USB           Registered USB driver for 28de:1205
D Mapper        Creating virtual devices
D Mapper        Keyboard: <scc.uinput.Keyboard object at 0x7f8bcd5a9610>
D Mapper        Mouse:    <scc.uinput.Mouse object at 0x7f8bcd5a9f90>
D Mapper        Gamepad:  <scc.uinput.UInput object at 0x7f8bcd5b5450>
D SCCDaemon     Created control socket /home/player/.config/scc/daemon.socket
D SCCDaemon     Connected to XServer :0.0
D asyncio       Using selector: EpollSelector
D deck          Got SteamDeck with serial 3:3
D USB           USB device added: 28de:1205
D AS  Mapper        Creating virtual devices
D AS  Mapper        Keyboard: <scc.uinput.Dummy object at 0x7f09f7771dd0>
D AS  Mapper        Mouse:    <scc.uinput.Dummy object at 0x7f09f740ecd0>
D AS  Mapper        Gamepad:  <scc.uinput.Dummy object at 0x7f09f740ed50>
D AS  AutoSwitcher  Parsed 30 autoswitcher conditions
D AS  AutoSwitcher  AutoSwitcher started
D AS  AutoSwitcher  Connected to daemon, version 0.4.8.21
I SCCDaemon     Registered scc-autoswitch-daemon
D AS  AutoSwitcher  Enabled: False
D AS  AutoSwitcher  Daemon reported profile change: /home/player/.config/scc/profiles/Desktop.sccprofile
D SCCDaemon     Assigned default_mapper to <Deck deck3:3>
D SCCDaemon     Turning gyrosensor ON
D SCCDaemon     Controller added: <Deck deck3:3>
D AS  AutoSwitcher  Enabled: True
D AS  AutoSwitcher  Daemon reported profile change: /home/player/.config/scc/profiles/Desktop.sccprofile
I CemuHook      Created CemuHookUDP Motion Provider
D AS  AutoSwitcher  Window switched: 56623107
W AS  Actions       Action ChangeProfileAction can't handle button press event
D OSD DaemonCtrl    Connected to daemon, version 0.4.8.21
D OSD DaemonCtrl    Daemon reported profile change for deck3:3: /home/player/.config/scc/profiles/Desktop.sccprofile
D OSD DaemonCtrl    Daemon is ready.
I SCCDaemon     Registered scc-osd-daemon
I OSD osd.daemon    Sucessfully registered as scc-osd-daemon
[..... system goes sleeping and wake up .... ]
Exception ignored on calling ctypes callback function: <bound method USBTransfer.__callbackWrapper of <scc.lib.usb1.USBTransfer object at 0x7f8bcd5bbf90>>
Traceback (most recent call last):
  File "/tmp/.mount_sc-conBfqwtA/usr/lib/python/scc/lib/usb1.py", line 321, in __callbackWrapper
    callback(self)
  File "/tmp/.mount_sc-conBfqwtA/usr/lib/python/scc/drivers/usb.py", line 47, in callback_wrapper
    transfer.submit()
  File "/tmp/.mount_sc-conBfqwtA/usr/lib/python/scc/lib/usb1.py", line 721, in submit
    raiseUSBError(result)
  File "/tmp/.mount_sc-conBfqwtA/usr/lib/python/scc/lib/usb1.py", line 117, in raiseUSBError
    raise STATUS_TO_EXCEPTION_DICT.get(value, USBError)(value)
scc.lib.usb1.USBErrorIO: LIBUSB_ERROR_IO [-1]
```

C0rn3j, 2024-10-28:
Reopen if you can reproduce it on the 0.4.9+ branch.

0.4.8 has a hacked-in libUSB (`scc/lib/usb1.py`) which is some random ancient version of libUSB.

0.4.9 uses libUSB from the env/system.

closed 2024-10-28 by C0rn3j

bell07, 2024-10-28:
Same issue with 0.4.9.7

```
player@game-deck ~/Downloads $ ./sc-controller-0.4.9.7-bookworm-x86_64.AppImage daemon debug
D SCCDaemon     Starting SCCDaemon...
D SCCDaemon     Initializing drivers...
W SCCDaemon     Skipping disabled driver 'ds4drv'
W SCCDaemon     Skipping disabled driver 'ds5drv'
W SCCDaemon     Skipping disabled driver 'fake'
W SCCDaemon     Skipping disabled driver 'remotepad'
D USB           Registered USB driver for 28de:1102
D USB           Registered USB driver for 28de:1142
D USB           Registered USB driver for 28de:1205
D Mapper        Creating virtual devices
D Mapper        Keyboard: <scc.uinput.Keyboard object at 0x7f6a23d25e10>
D Mapper        Mouse:    <scc.uinput.Mouse object at 0x7f6a23d271d0>
D Mapper        Gamepad:  <scc.uinput.UInput object at 0x7f6a23d3a090>
D SCCDaemon     Created control socket /home/player/.config/scc/daemon.socket
D SCCDaemon     Connected to XServer :0.0
Exception in thread Thread-2 (_threaded):
Traceback (most recent call last):
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
Exception in thread Thread-3 (_threaded):
Traceback (most recent call last):
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
    self.run()
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/threading.py", line 975, in run
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/site-packages/scc/sccdaemon.py", line 1443, in _threaded
    self._target(*self._args, **self._kwargs)
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/site-packages/scc/sccdaemon.py", line 1443, in _threaded
    self.p = subprocess.Popen(self.args, stdin=None)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/subprocess.py", line 1024, in __init__
    self.p = subprocess.Popen(self.args, stdin=None)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/subprocess.py", line 1024, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/subprocess.py", line 1901, in _execute_child
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/subprocess.py", line 1901, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/player/Downloads/usr/bin/python'
FileNotFoundError: [Errno 2] No such file or directory: '/home/player/Downloads/usr/bin/python'
D asyncio       Using selector: EpollSelector
D deck          Got SteamDeck with serial 3:3
D USB           USB device added: 28de:1205
D root          TODO: Hardcoded 1s sleep!
D SCCDaemon     Assigned default_mapper to <Deck deck3:3>
D SCCDaemon     Turning gyrosensor ON
D SCCDaemon     Controller added: <Deck deck3:3>
I CemuHook      Created CemuHookUDP Motion Provider
[..... system goes sleeping and wake up .... ]
Exception ignored on calling ctypes callback function: <bound method USBTransfer.__callbackWrapper of <class 'usb1.USBTransfer'>>
Traceback (most recent call last):
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3/dist-packages/usb1/__init__.py", line 328, in __callbackWrapper
    callback(self)
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3.11/site-packages/scc/drivers/usb.py", line 57, in callback_wrapper
    transfer.submit()
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3/dist-packages/usb1/__init__.py", line 748, in submit
    raiseUSBError(result)
  File "/tmp/.mount_sc-congtYQW1/usr/lib/python3/dist-packages/usb1/__init__.py", line 120, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorIO: LIBUSB_ERROR_IO [-1]
D root          TODO: Hardcoded 1s sleep!
```