C0rn3j/sc-controller #58: Issues with build and run the package byself

bell07, 2024-10-29:
I try to follow the steps in README to build and use local copy for testing reason without installation.
This issue documents my steps and caveats, maybe you can explain more in readme

**Building the package by yourself**
Dependencies installed (I think so). Some packages I did not found in Gentoo portage:
dev-python/libusb1 is in guru overlay only
ioctl-opt is not available trough package manager, therefore installed using pip.

```
$ python3 -m build --wheel
[... lot of output ...]
Successfully built sccontroller-0.4.9.9.dev1+g6254f42-cp312-cp312-linux_x86_64.whl

$ python3 -m installer --destdir="./build" dist/*.whl
[... no output ...] 

$ ./build/usr/bin/sc-controller
Traceback (most recent call last):
  File "/home/player/Projekte/sc-controller/./build/usr/bin/sc-controller", line 5, in <module>
    from scc.bin.sc_controller import main
ModuleNotFoundError: No module named 'scc'
```

**Via Python venv through run.sh**
```
$ ./run.sh
x86_64-linux-gnu-gcc not found, install it. The package is usually named gcc!
```
gcc is installed, but the binary is called x86_64-pc-linux-gnu-gcc, or just gcc

```
--- a/run.sh
+++ b/run.sh
@@ -46,8 +46,8 @@ function testDeps() {
 #              echo -e "${Red}gi.Gtk not found, install it. ${Yellow}The package may be named gtk3 or gir1.2-gtk-3.0 on your distribution!${NoColor}"
 #              exit 1
 #      fi
-       if ! command -v x86_64-linux-gnu-gcc >/dev/null; then
-               echo -e "${Red}x86_64-linux-gnu-gcc not found, install it. ${Yellow}The package is usually named gcc!${NoColor}"
+       if ! command -v gcc >/dev/null; then
+               echo -e "${Red}gcc not found, install it. ${Yellow}The package is usually named gcc!${NoColor}"
                exit 1
        fi
 }
```

Next try:

```
$ ./run.sh
[... lot of output ...]
removing build/bdist.linux-x86_64/wheel
Successfully built sccontroller-0.4.9.9.dev1+g6254f42.d20241029-cp312-cp312-linux_x86_64.whl
Processing ./dist/sccontroller-0.4.9.9.dev1+g6254f42.d20241029-cp312-cp312-linux_x86_64.whl
sccontroller is already installed with the same version as the provided wheel. Use --force-reinstall to force an installation of the wheel.

[notice] A new release of pip is available: 24.2 -> 24.3.1
[notice] To update, run: pip install --upgrade pip
./run.sh: Zeile 75: sc-controller: Command not found.
```

got it run using 
```
$ PATH="$PWD/build/usr/bin:$PATH" ./run.sh
```
The path should be adjusted in run.sh

**Now the launch appears**
.. but the daemon is not running

```
$ PATH="$PWD/build/usr/bin:$PATH" ./run.sh daemon debug
[... lot of output ...]
D SCCDaemon     Starting SCCDaemon...
D SCCDaemon     Initializing drivers...
Traceback (most recent call last):
  File "/home/belka/Projekte/sc-controller/build/usr/bin/scc-daemon", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/belka/Projekte/sc-controller/scc/bin/scc_daemon.py", line 34, in main
    daemon.debug()
  File "/home/belka/Projekte/sc-controller/scc/sccdaemon.py", line 1133, in debug
    self.run()
  File "/home/belka/Projekte/sc-controller/scc/sccdaemon.py", line 663, in run
    self.init_drivers()
  File "/home/belka/Projekte/sc-controller/scc/sccdaemon.py", line 93, in init_drivers
    mod = getattr(__import__('scc.drivers.%s' % (modname,)).drivers, modname)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/belka/Projekte/sc-controller/scc/drivers/ds4drv.py", line 19, in <module>
    from scc.drivers.hiddrv import (
  File "/home/belka/Projekte/sc-controller/scc/drivers/hiddrv.py", line 203, in <module>
    _lib = find_library("libhiddrv")
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/belka/Projekte/sc-controller/scc/tools.py", line 342, in find_library
    raise OSError('Cant find %s.so. searched at:\n %s' % (libname, '\n'.join(search_paths)))
OSError: Cant find libhiddrv.so. searched at:
 /home/belka/Projekte/sc-controller/libhiddrv.cpython-312-x86_64-linux-gnu.so

```
Hacky additional patch

```
--- a/scc/tools.py
+++ b/scc/tools.py
@@ -328,7 +328,7 @@ def find_library(libname):
        for extension in so_extensions:
                search_paths += [
                        os.path.abspath(os.path.normpath(
-                               os.path.join( base_path, '..', libname + extension ))),
+                               os.path.join( base_path, '..', "build", "usr/lib/python3.12/site-packages", libname + extension ))),
                        os.path.abspath(os.path.normpath(
                                os.path.join( base_path, '../..', libname + extension )))
                        ]
```

Now I am able to test the sc-controller ;-)
I'm sure it should be easier

C0rn3j, 2024-10-29:
> gcc is installed, but the binary is called x86_64-pc-linux-gnu-gcc, or just gcc

Arch Linux symlinks it without `-pc` because Rust. Switched to the regular version.

https://gitlab.archlinux.org/archlinux/packaging/packages/gcc/-/commit/e34a6d09ea400ec269fd764e45a3b000af008719

> I'm sure it should be easier

We've majorly switched tooling in the beta, so it's a bit rough still.

I've also apparently tested things while having the system version installed so it kept falling back on it.

I've fixed run.sh and added site-package search of the current env.

Manual instructions still need a bit of fixing, mainly due to the .so modules again.

C0rn3j, 2024-10-31:
I've fixed local builds, so everything should be taken care of.

https://github.com/C0rn3j/sc-controller/commit/a856067b763294753d376dcfb5a32c0abac4de41

closed 2024-10-31 by C0rn3j