C0rn3j/sc-controller #62: Thank you

Saeniv, 2025-01-03:
sc-controller does not detect my DS4 controller. Last year I was using it for a game and it worked perfectly via bluetooth and cable. Now I can see it under register new controller, which asks me to restart emulation, but won't detect the controller neither by cable nor Bluetooth. So it kind of detect it, but does not show the DS4 picture. 

I have read through some issues and I guess it is somehow related to them. 

Thank you very much for your work and time.

C0rn3j, 2025-01-03:
Bluetooth won't work by default due to https://github.com/C0rn3j/sc-controller/issues/22

Cable should work but maybe only on the beta version - are you using AUR/sc-controller? If so, try AUR/sc-controller-git.

I just haven't had the time to polish it up for a release yet.

Saeniv, 2025-01-03:
You are right, the AUR/sc-controller-git version does detect the controller via USB. 
I was just trying the BETA version, my mistake. 

Thank you for providing help and the information about issue 22.

C0rn3j, 2025-01-03:
Glad to hear you got it working

closed 2025-01-03 by C0rn3j