C0rn3j/sc-controller #63: Lenovo Legion Go Controller Support?

Kreyren, 2025-01-14:
Cross-Referencing: https://github.com/Arcanyx-org/NiXium/pull/156

The controllers seems to be a standard **Xbox One Elite** Controller connected via USB in a fancy form-factor so everything works out of the box in sc-controller with `xpadneo` driver, but the issue is that if i add the controllers as xbox 360 controller (there isn't any better option) it won't let me map the paddles. -> Could you maybe add config in  https://github.com/C0rn3j/sc-controller/blob/bacf4194bade7a2e7483c15d3a01c007b1865a0c/glade/creg.glade#L58 that would let me assign those paddles?

The HandHeld Daemon won't let me map the controls so sc-controller would be a preferable choice.

Kreyren, 2025-01-14:
[image]

I've made the controller reference in GIMP in case it helps, i tried to follow the sc-controller style

[lengo_sc_controller.xcf.zip](https://github.com/user-attachments/files/18404793/lengo_sc_controller.xcf.zip) (remove the .zip suffix//)

C0rn3j, 2025-01-14:
Hey there - thanks for the information and design!

I am of course not against adding support for new devices, but have no experience with doing so, lack the device for testing and currently also lack much time to put towards SCC.

If you or someone else wanted to PR the support, I would suggest to look at the commits/PR that added Steam Deck.