C0rn3j/sc-controller #67: AppImage OSD-KEYBOARD fails in Linux Mint

jujlguljg, 2025-01-25:
**I can not tell you every detail accurately.**

./appimage osd-keyboard
`/tmp/.mount_sc-conVpeBgW/usr/lib/python3.12/site-packages/scc/osd/__init__.py:11: PyGIWarning: Gdk was imported without specifying a version first. Use gi.require_version('Gdk', '3.0') before import to ensure that the right version gets loaded.
  from gi.repository import Gdk, GLib, GObject, Gtk
Gtk-Message: 21:40:27.961: Failed to load module "xapp-gtk3-module"
Traceback (most recent call last):
  File "/tmp/.mount_sc-conVpeBgW/usr/bin/scc", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/tmp/.mount_sc-conVpeBgW/usr/lib/python3.12/site-packages/scc/scripts.py", line 397, in main
    sys.exit(command(sys.argv[0], sys.argv[2:]))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_sc-conVpeBgW/usr/lib/python3.12/site-packages/scc/scripts.py", line 90, in cmd_osd_keyboard
    return run_osd_tool(Keyboard(), argv0, argv)
                        ^^^^^^^^^^
  File "/tmp/.mount_sc-conVpeBgW/usr/lib/python3.12/site-packages/scc/osd/keyboard.py", line 318, in __init__
    OSDWindow.__init__(self, "osd-keyboard")
  File "/tmp/.mount_sc-conVpeBgW/usr/lib/python3.12/site-packages/scc/osd/__init__.py", line 59, in __init__
    gi.require_version("GtkLayerShell", "0.1")
  File "/tmp/.mount_sc-conVpeBgW/usr/lib/python3/dist-packages/gi/__init__.py", line 122, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace GtkLayerShell not available`

**Gtk-Message: 21:40:27.961: Failed to load module "xapp-gtk3-module"**
when trying to sudo apt install libxapp-gtk3-module
`trying to overwrite '/etc/X11/Xsession.d/80xapp-gtk3-module', which is also in package libxapp1:amd64 2.8.8+xia`

someone in Mint IRC said libxapp1 has xapp-gtk3-module in it
we verified with:
ls -la /usr/lib/x86_64-linux-gnu/gtk-3.0/modules/libxapp-gtk3-module.so 
`-rw-r--r-- 1 root root 14328 Jan  6 09:17 /usr/lib/x86_64-linux-gnu/gtk-3.0/modules/libxapp-gtk3-module.so`

i think this is because you are packaged for ubuntu and not mint.
mint made xapps iirc and have messed with something

thats all i can tell you

libgtk-layershell0 is installed

C0rn3j, 2025-01-25:
Are you using GNOME? 

Mutter does not support LayerShell.

jujlguljg, 2025-01-26:
> Are you using GNOME?
Xfce.

**On a Fresh Cinnamon and Xfce VM, 0.5.0 has the same problem.**
x11 btw.

jujlguljg, 2025-01-26:
here is the traceback
idk

`  File "/tmp/.mount_sc-conBOAtT4/usr/bin/scc", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/tmp/.mount_sc-conBOAtT4/usr/lib/python3.12/site-packages/scc/scripts.py", line 397, in main
    sys.exit(command(sys.argv[0], sys.argv[2:]))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_sc-conBOAtT4/usr/lib/python3.12/site-packages/scc/scripts.py", line 89, in cmd_osd_keyboard
    from scc.osd.keyboard import Keyboard
  File "/tmp/.mount_sc-conBOAtT4/usr/lib/python3.12/site-packages/scc/osd/__init__.py", line 14, in <module>
    gi.require_version("GtkLayerShell", "0.1")
  File "/tmp/.mount_sc-conBOAtT4/usr/lib/python3/dist-packages/gi/__init__.py", line 122, in require_version
`

C0rn3j, 2025-01-26:
The package with the bindings on Ubuntu 24.04 is `gir1.2-gtklayershell-0.1`, not sure about Mint.

```python
[0] # python
>>> import gi
>>> gi.require_version("GtkLayerShell", "0.1")
```

You should be able to import this without errors.

PoorPocketsMcNewHold, 2025-02-14:
The same goes for Nobara 41 (Fedora based) on KDE Plasma. And yes, I had `gtk-layer-shell-0.9.0-1.fc41.x86_64` installed. If I remember, version 0.4.9 worked.
```
(sc-controller:33049): Gtk-WARNING **: 10:14:55.512: Theme parsing error: gtk.css:2:0: expected semicolon
Traceback (most recent call last):
  File "/tmp/.mount_sc-conSzhbL4/usr/bin/sc-controller", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/tmp/.mount_sc-conSzhbL4/usr/lib/python3.12/site-packages/scc/bin/sc_controller.py", line 30, in main
    from scc.gui.app import App
  File "/tmp/.mount_sc-conSzhbL4/usr/lib/python3.12/site-packages/scc/gui/app.py", line 29, in <module>
    import scc.osd.menu_generators
  File "/tmp/.mount_sc-conSzhbL4/usr/lib/python3.12/site-packages/scc/osd/__init__.py", line 14, in <module>
    gi.require_version("GtkLayerShell", "0.1")
  File "/tmp/.mount_sc-conSzhbL4/usr/lib/python3/dist-packages/gi/__init__.py", line 122, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace GtkLayerShell not available
```

jujlguljg, 2025-03-27:
> gtk-layer-shell-0.9.0-1.fc41.x86_64 installed. Previous version 0.4.9.8
> Same problem on the same kind of system

Just saying hi to us other lucky hacker steam controller users x)

Thx for maintaining