C0rn3j/sc-controller #72: Daemon fails to start with libusb1 3.3.0

chewi, 2025-03-06:
It works with 3.2.0, but with 3.3.0, `scc daemon debug` crashes:

```
D SCCDaemon     Starting SCCDaemon...
D SCCDaemon     Initializing drivers...
W SCCDaemon     Skipping disabled driver 'ds4drv'
W SCCDaemon     Skipping disabled driver 'fake'
W SCCDaemon     Skipping disabled driver 'hiddrv'
W SCCDaemon     Skipping disabled driver 'remotepad'
D USB           Registered USB driver for 054c:0ce6
D USB           Registered USB driver for 28de:1102
D USB           Registered USB driver for 28de:1142
D USB           Registered USB driver for 28de:1205
D Mapper        Creating virtual devices
D Mapper        Keyboard: <scc.uinput.Keyboard object at 0x7f09dd1bd940>
D Mapper        Mouse:    <scc.uinput.Mouse object at 0x7f09dd1bda90>
D Mapper        Gamepad:  <scc.uinput.UInput object at 0x7f09dd1bdbe0>
D SCCDaemon     Created control socket /home/chewi/.config/scc/daemon.socket
W SCCDaemon     Wayland with WLRoots detected. Disabling X11 support, some functionality will be unavailable
Traceback (most recent call last):
  File "/usr/bin/scc-daemon", line 8, in <module>
    sys.exit(main())
             ~~~~^^
  File "/usr/lib/python3.13/site-packages/scc/bin/scc_daemon.py", line 34, in main
    daemon.debug()
    ~~~~~~~~~~~~^^
  File "/usr/lib/python3.13/site-packages/scc/sccdaemon.py", line 1133, in debug
    self.run()
    ~~~~~~~~^^
  File "/usr/lib/python3.13/site-packages/scc/sccdaemon.py", line 673, in run
    self.start_drivers()
    ~~~~~~~~~~~~~~~~~~^^
  File "/usr/lib/python3.13/site-packages/scc/sccdaemon.py", line 129, in start_drivers
    s(self)
    ~^^^^^^
  File "/usr/lib/python3.13/site-packages/scc/drivers/usb.py", line 353, in start
    _usb.start()
    ~~~~~~~~~~^^
  File "/usr/lib/python3.13/site-packages/scc/drivers/usb.py", line 231, in start
    self._ctx.setPollFDNotifiers(register_fd, unregister_fd)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/usb1/__init__.py", line 2205, in wrapper
    return func(self, *args, **kw)
  File "/usr/lib/python3.13/site-packages/usb1/__init__.py", line 2606, in setPollFDNotifiers
    self.getFinalizer(
    ^^^^^^^^^^^^^^^^^
AttributeError: 'USBContext' object has no attribute 'getFinalizer'. Did you mean: '_getFinalizer'?
E OSD osd.daemon    Connection to daemon lost                                                   
```

I don't know whether this is a problem with sc-controller or libusb1.

C0rn3j, 2025-03-21:
Could you check if hacking libusb1 like https://github.com/vpelletier/python-libusb1/pull/107 fixes it?

chewi, 2025-03-21:
Yes, I can confirm that fixes it.

C0rn3j, 2025-03-24:
Thanks for testing, the PR was pulled and this is now fixed.

closed 2025-03-24 by C0rn3j