C0rn3j/sc-controller #80: [Bug] Steam Deck Control Emulation Stops When Device Sleeps

Jazztache, 2025-05-18:
Hello!

I am using SC-Controller on my Steam Deck running the latest build of Arch Linux and KDE.
I am running sc-controller 0.5.1-1 from the AUR.
The Deck is a standard LCD deck.

When I put the device to sleep, control emulation stops.
I unlock my screen and none of the buttons work.

I can alleviate this by:
- Tapping the icon on the systray
- Pressing 'Stop Emulation'
- Wait a bit
- Press 'Start Emulation'
- Wait a bit more
- Emulation is back.

This is very annoying and I would like for emulation to resume when sleep is over seamlessly.

I am not sure if this is an issue with the daemon or application itself.
If there are any commands I can use to debug this issue, be it from `hidraw` or from `scc` itself, please tell me.

Thankyou!

C0rn3j, 2025-05-18:
Possible dupe of #57, you can verify by launching the daemon in debug mode and checking if the issue is the same

bell07, 2025-07-02:
From my point of view it looks like the same issue I have on Gentoo. You can try to fix it using elogind script:

cat /etc/elogind/system-sleep/steamdeck-sleep.sh 

```
#!/bin/sh
case $1 in
   pre)
     su -c 'DISPLAY=:0 /usr/bin/scc-daemon stop --once' player
     ;;

   post)
     su -c 'DISPLAY=:0 /usr/bin/scc-daemon restart' player &
     ;;
esac

exit 0
```

On my system the restart freezes my Gentoo on resume. But maybe the Arch is able to wake up properly

It the script does not work at all:
The script is for Xorg (DISPLAY=:0)
Check if the scc-daemon is the same on your system.
Maybe the path could be /etc/systemd/system-sleep. Check Arch docs

Let me know if your system freezes or not. If not I start to search for the reason